ERFSEDS/common_modules #5: Voltage Regulation

NathanRoseCE, 2020-06-03:
Circuit Voltage Regulation

ckeer0, 2020-07-13:
Minor Issues: 

I can't for the life of me view your footprints. I think that it may just be me, but I just can't.

3.3V needs a power flag 

PCB has weird traces on the shield for the USB-C. 
Silkscreen needs some serious work. 

Functionally, it works, it's just the little things.

NathanRoseCE, 2020-07-14:
Create one library that contains for each module and has all parts needed for the module
3.3V does not need a power flag @ckeer0 as the symbols will be power output and so a power flag will cause an output connected to output warning
Each module should only have the libary needed for itself

ckeer0, 2020-07-24:
Issue closed

closed 2020-07-24 by ckeer0