ERFSEDS/common_modules #6: Microprocessor Development

NathanRoseCE, 2020-06-03:
Create the module for the microprocessor

ckeer0, 2020-07-01:
Requesting Review of Module

NathanRoseCE, 2020-07-02:
Schematic
Update Schematic Sheet to have Cerberus Sheet layout
Library name has ' ' which does not work with all versions of kicad change to '-' or '_'

Layout
Unnecessary via by header pin PD2
Unnecessary via left of header PC11
Reset trace has Unnecessary spike
Vusb has Unnecessary loop antenna(minor issue)
Bottom Left of MCU footprint has some ugly GND traces(minor issue)
Unnecessary traces above C906(minor issue)
MCU pin 32 trace has an unnecessary and easily fixable right angle(minor issue)

NathanRoseCE, 2020-07-03:
Datasheet Required(not going to put you back to in progress for that but will not mark as complete
missing signature from Cerberus LEad

ckeer0, 2020-07-24:
Issue closed

closed 2020-07-24 by ckeer0