ERFSEDS/common_modules #7: Flash Development

NathanRoseCE, 2020-06-03:
Flash chip needs to be developed

ckeer0, 2020-07-06:
Rename CLK on SD to SCLK

Place Power flags on 3v3 and GND to indicate power inputs (not necessary but good design practice. Plus it eliminates warnings on the kicad ERC) 

This could be me, but I don't know if your symbol library is formatted correctly. When I added it it still gives me blank symbols and wants to put no connect pins everywhere

For the pcb, idk if its just a personal thing, or if its a design thing, but Idk if I like the mosi splitting off like that before the pad. Maybe try routing directly from the pin to the pad, and then the pad to the SD pad. Same thing with MISO. Again, could just be me. Nathan can confirm or deny lol.

NathanRoseCE, 2020-07-08:
The readme tells me nothing about how to use it, and just adds confusion

ckeer0, 2020-07-24:
Issue closed

closed 2020-07-24 by ckeer0